Infineon/AIROC-Wi-Fi-Bluetooth-STM32 #1: Please add support for CYW43455

spiralray, 2021-05-29:
# Is your feature request related to a problem? Please describe.

Hi, now I'm trying to make [1MW M.2 Module](https://www.embeddedartists.com/products/1mw-m-2-module/) work on STM32H747 Discovery Kit.

CYW43455 supports 802.11 ac on 3.3V environment.
So I think it's more convenient than CYW43012. (Hardware modification is not needed!)

Now, I ported the parameters and the firmware of CYW43455  from [iMX RT1176 Developer’s Kit](https://www.embeddedartists.com/products/imx-rt1176-developers-kit/) to this project.

With the modified project, STM32H7+CYW43455 successfully showed scan results of wifi_scan,  but `cyhal_sdio_bulk_transfer` returned `HAL_SD_ERROR_CMD_CRC_FAIL`.

```
******************* WiFi-Scan app ******************* 

    Push blue button or send any symbol via serial terminal to continue…

    CYW43xxx detected

whd_bus_sdio_cmd53:874 cyhal_sdio_bulk_transfer SDIO_BYTE_MODE failed
Function whd_bus_sdio_read_backplane_value failed at line 735 checkres = 33555501 
WLAN MAC Address : -------------------
WLAN Firmware    : wl0: Nov 13 2019 04:39:05 version 7.45.198 (r723338 CY) FWID 01-369a6798
WLAN CLM         : API: 12.2 Data: 9.10.136 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2019-05-02 02:31:24 
WHD VERSION      : v1.93.0 : v1.93.0 : GCC 9.3 : 2020-12-21 13:24:03 +0530

—————————————————————————————————————————————————
  #                  SSID                  RSSI   Channel       MAC Address              Security
——————————————————————————————————————————————————
  1  -------------                       -56      36      -------------         WPA2_AES_PSK   
  2   -------------                        -79       1      -------------         WPA2_AES_PSK   

...

 16   -------------                         -50     116      -------------         WPA2_AES_PSK   
SDIO bus failed to come up , whd_ensure_wlan_bus_is_up failed at 926 
Could not bring bus back upFunction whd_wifi_set_ioctl_value failed at line 3106 checkres = 101580800 
SDIO bus failed to come up , whd_ensure_wlan_bus_is_up failed at 926 
Could not bring bus upSDIO bus failed to come up , whd_ensure_wlan_bus_is_up failed at 926 
Could not bring bus back upFunction whd_management_set_event_handler failed at line 342 checkres = 101580800 
Function whd_wifi_scan failed at line 2266 checkres = 101580800 
SDIO bus failed to come up , whd_ensure_wlan_bus_is_up failed at 926 
Could not bring bus back upFunction whd_wifi_set_ioctl_value failed at line 3106 checkres = 101580800 
SDIO bus failed to come up , whd_ensure_wlan_bus_is_up failed at 926 
Could not bring bus upSDIO bus failed to come up , whd_ensure_wlan_bus_is_up failed at 926 
Could not bring bus back upFunction whd_management_set_event_handler failed at line 342 checkres = 101580800 
Function whd_wifi_scan failed at line 2266 checkres = 101580800 
SDIO bus failed to come up , whd_ensure_wlan_bus_is_up failed at 926 
...
```

In my opinion, this is because the firmware which I used is not for WHD but for WICED. (Is that correct?)

# Describe the solution you'd like

Could you add support for CYW43455?

(I'm sorry for my poor English.)

spiralray, 2021-05-29:
I'm sorry, I fix the issue.
I mistakenly set the PMU base address, and `save_restore_enable` got FALSE value.

It worked fine when "if" statements set completely same as CYW43012 in `whd_chip.c`.

But even now, I hope official support of CYW43455.

closed 2021-06-08 by spiralray

yolominwoo, 2024-04-08:
@spiralray Can you give me a guide for how to solve?