Infineon/AIROC-Wi-Fi-Bluetooth-STM32 #2: Wi-Fi Direct example

jeanbie0, 2021-07-12:
**Have you looked for an answer?**

Yes without any success.

**Question/Issue**:

Is this pack compatible with Wi-Fi Direct ? (In group owner mode).
If it's compatible, do you have some examples?
Thank you.

**Context**

STM32H723 with Laird Sterling LWB 450-0152 (CYW 4343W).
Example 'WiFi BLE onboarding' is working

###### THANKS!

rmbk19, 2021-07-13:
Hi @jeanbie0, 

The issue is currently being handled in this thread -  [Wi-Fi direct in STM32 connectivity pack with WHD.](https://community.cypress.com/t5/Wireless/Wi-Fi-direct-in-STM32-connectivity-pack-with-WHD/td-p/282627) 

Regards, 
Rakshith

Denzerek, 2021-11-19:
Hi @jeanbie0,
Unfortunately we do not have Wifi Direct implemented as of now  as well as the supporting libraries in either the Anycloud SDK or any other platforms running on Wifi Direct.

Please contact local sales/marketing team to get an estimated timeline as to when this might get implemented.

Thanks