Infineon/AIROC-Wi-Fi-Bluetooth-STM32 #5: using the source-code of stm32-connectivity pack

aniruddhann, 2022-03-17:
**Have you looked for an answer?**

I had looked on the community support portal & gone through the License.txt file. 

**Question/Issue**:

Can you let me know, if I can use the STM32-connectivity pack source code and it's different components (like WHD, WCM, etc) by modification/addition of code from licensing point of view?

**Context**

I am planning to port this connectivity pack source code on the STM32 chip to communicate with muRata(CYW4343). I am using Zephyr framework/RTOS, so, I need to modify/add the source code accordingly.
So, Kindly let me know if it's permissible? what type of licensing this has?

**Where did you expect to find the answer?**

###### THANKS!

Aditi1091998, 2022-03-28:
Hi,

Yes, you can go ahead and use the Zephyr framework with the connectivity pack source code.

Thanks

Aditi1091998, 2022-03-28:
With regards to license, you can modify the driver (WHD) as per your use case framework as long as the modifications enable the use of a Cypress/Infineon hardware product on a particular host operating system and in your case the wifi chip is CYW4343W so there are no issues.

Thanks

ifyall, 2022-04-12:
Closing as comments from Aditi above are accurate.

closed 2022-04-12 by ifyall