Infineon/AIROC-Wi-Fi-Bluetooth-STM32 #6: Performance between Murata Type 1lv (CYW43012) and STM32H747-DISCO

danix2k, 2022-06-01:
**Question/Issue**:
Performance between Murata Type 1lv (CYW43012) and STM32H747-DISCO

**Context**
I'm evaluating the WIFI/BLE module 1LV by Murata with a STM32H747-Dicovery board and Murata adapter as reported in the PDF file in https://github.com/Infineon/stm32-connectivity/tree/master/Documentation

The connection works fine (wifi scan OK and connection OK) and now I'm trying to increase the communication performance; the first test, using LWIP stack as server on MCU and ipref client on windows/linux side gave me only 0.15-0.20MB/s (150kBps) that seems to me very VERY slow using SDIO interface (with 25MHz clock). 

Do anybody have any clues to increase this bitrate?

I also tested UDP throughput and it is around 0.60-0.65MB/s (about 5Mbps).

The big point here is the following: I have the same performance setting CPU CLOCK at 75MHz or 400MHz and setting SDIO CLK to 25MHz to 50MHz. Same result in both test case.

So, it seems to me that is a driver/SW problem in the stm32-connectivity pack.

Has anyone TCP/UDP throughput data result with FreeRTOS and STM32 connectivity pack?

Thanks

ifyall, 2022-06-28:
@danix2k, thank you for pointing this out. We will be taking a look at this starting this week and will post updates when they are available.

Ian

closed 2024-02-27 by ifyall

jenish-rudani, 2025-12-12:
Where is this issue referred to? 

I am unable to increase this throughput above 250KBps