Infineon/AIROC-Wi-Fi-Bluetooth-STM32 #11: Example bug request

2Bumsu, 2024-01-03:
hello.

I purchased an STM32H7 module and a 2BC WiFi M.2 module.

I'm trying to run the example.

The previously posted examples were not written based on CYW4373.

So, I recreated the project with STM32 CubeMX and created it both with CUBE IDE and EWARM.

But all build errors occur.

Could you please make at least one STM32H7 DisCovery + CYW4373 WiFi SCAN project?

please.

ZenoMaverick, 2024-01-03:
Hey @2Bumsu, CYW4373 (Type2BC) is supported in this expansion pack.
Have you taken a look at the [user guide](https://github.com/Infineon/AIROC-Wi-Fi-Bluetooth-STM32/blob/master/Documentation/InfineonAIROC-Wi-Fi-Bluetooth-STM32ExpansionPack_UserGuide.pdf)?

Under section 7.2.1, you'll find the instructions to change the module part to build the project for 2BC.

ZenoMaverick, 2024-01-04:
@2Bumsu According to user guide section 1.5.1, IAR needs 9.30.1.
Do try it again with the compatible IDE version.

npal-cy, 2024-01-08:
Hi @2Bumsu , could you try to make path length shorter.
looks like " "C:\Users\icomg\Desktop\AIROC-Wi-Fi-Bluetooth-STM32-master\Projects\STM32H747I-DISCO\Applications\wifi_scan\Middlewares\Third_Party\Infineon_Wireless_Connectivity\
btstack-integration\COMPONENT_HCI-UART\firmware\COMPONENT_4373\COMPONENT_MURATA-2BC\w_bt_firmware_controller.c" > 256 and IAR does not see that file. Please copy wifi_scan to some C:\tmp and check if it solve the problem.

2Bumsu, 2024-01-08:
hello. Thankfully it was resolved.

However, I ran the WiFi_SCAN example, but there was no response.

If I regenerate the project, will I have to rewrite my code?

npal-cy, 2024-01-08:
after change the module, and regenerate the project you need remove from workspace some src/includes related to previous module. Please refer to section 9.2 of User Guide for details (It should  be fixed in future version of STm32CubeMx).

2Bumsu, 2024-01-10:
It's sad because it's difficult to execute a project... I was trying to do it at work, but since I'm doing it at home, I'll give it a try and leave a reply. thank you