Infineon/AIROC-Wi-Fi-Bluetooth-STM32 #12: CMD52 Timeout for Type 1YN (CYW43439)

neeloza-embrill, 2024-02-17:
**Question/Issue**:

We are facing issue to interface Type 1YN (CYW43439) with STM32H7 MCU.

**Context**

I am developing an application using Wi-Fi. I am facing one issue to initialise WCM. (cy_wcm_init)

When it enable "F1 and F2" in whd_bus_sdio_protocol.c, it timeout to execute CMD52. 

Same issue I am facing with Type 1LV (CYW43012) and 1.5.1 & 1.5.0 release. 

Evaluation board MCU: NUCLEO-H723ZG
Wi-Fi Evaluation Board: 1LV M.2 Module, 1YN M.2 Module (Manufactured by "Embedded Artists")
Adaptor Board: uSD-M.2 Adapter (muRata )

###### THANKS!

neilch0811, 2024-02-19:
Hi,

Have you also consulted with MuRata? Post your question to 

https://community.murata.com/s/topic/0TO5F000002DCajWAG/stm32-nucleo-boards

And, if you run with NUCLEO-144 MCU kit, you should purchase the adapter board + 1YN or 1LV - https://www.murata.com/en-global/products/connectivitymodule/wi-fi-bluetooth/overview/stm32-nucleo

Let me know if you still have the question. 

Thanks.

npal-cy, 2024-02-19:
Hi @neeloza-embrill ,

Yes, agree with @neilch0811 that it is better to use  [stm32-nucleo adapter](https://www.murata.com/en-global/products/connectivitymodule/wi-fi-bluetooth/overview/stm32-nucleo) for STM32 Nucleo boards.

Possible issues:
- SDIO wired ribbon cable connection from uSD-M.2 Adapter to Nucleo board can have signal integrity issues.
>> you can try to override SDIO clock to 5Mhz by set in STM32CubeMx SDMMC_CLK_FREQ_OVERRIDE to 5000000 (in AIROC component setting).

- for 1LV M.2 Module required 1.8V for SDIO signal and 3.3V for control signal, so need 1.8V VDD configuration of STM32 Nucleo board or some specific level shifter. uSD-M.2 Adapter  doesn't provide level shifters on SDIO bus (only on UART). Type 1YN can works on  3.3V for SDIO, please refer to https://www.embeddedartists.com/wp-content/uploads/2021/05/1YN_M2_Datasheet.pdf (Section 3.5 VDDIO Override Feature )

We tested NUCLEO-H745ZI-Q + Murata stm32-nucleo adapter with 1LV M.2 Module.

Regards,
Nazar

ifyall, 2024-02-27:
@neeloza-embrill, have you been able to reach out to Murata and acquire one of their adapters? We want to make sure that you are able to move forward before resolving this ticket.

Thank you,

Ian

neeloza-embrill, 2024-02-28:
Hello @ifyall , Thanks for reaching out to me. As per @npal-cy suggestion, We have ordered "stm32-nucleo adaptor" for testing. Please give us few days to reach this adaptor board to our place. We will validate and get back to you ASAP.

neeloza-embrill, 2024-03-07:
Hello @npal-cy ,

We have tried to implement Type 1YN with STM32H753 NUCLEO board using STM32 Nucleo Adaptor (uSD-M.2 ). We have Tested with 1.8V and 3V3.

We are facing one issue to initialise WCM. (cy_wcm_init() function). When it enable "F1 and F2" in "whd_bus_sdio_protocol.c", it timeout to execute CMD52.
 
After overriding SDIO clock to 5Mhz by set in STM32CubeMx "SDMMC_CLK_FREQ_OVERRIDE", we are facing the same issue of WCM initialisation.
 
Jumper settings on STM32 Nucleo Adaptor is as per the [document](https://mm.digikey.com/Volume0/opasdata/d220001/medias/docus/5784/STM32%20Nucleo%20RTOS.pdf) provided by Murata.
 
Took reference from https://github.com/Infineon/AIROC-Wi-Fi-Bluetooth-STM32 (STM32H563) and successfully implemented for Type 1LV with STM32H753 NUCLEO board using STM32 Nucleo Adaptor (uSD-M.2 ).

We have also tested "Type 1YN" and "Type 1LV" using STM32 H747I-DISCO board over SDIO interface (like a SD-Card connector) and Wi-Fi is successfully working. So, we can say that Type 1YN hardware is also working.
 
Error logs when we tried Type 1YN with 1.8V:
 
> WiFi-Scan app (FreeRTOS V10.3.1) 
> [2024-03-05 19:33:12] 
> [2024-03-05 19:33:14] Timeout while waiting for function 2 to be ready
> [2024-03-05 19:33:14] Interface is not up/NULL and failed in function whd_deinit at line 193 
> [2024-03-05 19:33:14] Failed to initialize Wi-Fi
> [2024-03-05 19:33:41] Timeout while waiting for function 2 to be ready
> [2024-03-05 19:33:41] Interface is not up/NULL and failed in function whd_deinit at line 193 
> [2024-03-05 19:33:41] Failed to initialize Wi-Fi

npal-cy, 2024-03-07:
Hi @neeloza-embrill,

Please double check that after switch  Type 1YN, the project does not have any source/include to previous selected Type 1LV. STM32CubeMx has a problem, it does not remove sources/includes of the PDSC component from the project workspace (STM32CubeIde/EWARM), when another variant of this component is disabled or changed. We added this behavior i Errata section of UG: 
[image]

also Section 9.2 'Workaround when changing device/module' detailed describe the issue..

If above information does not resolve the problem, please share your project (with generated Middlewares folder and build outputs..) i will check and try on my side.

Regards,
Nazar

prachichechani, 2024-03-11:
Hi, 

Still getting error in WCM initialisation. 

Respective Project is attached below :

[WiFi_Type_1yn_STM32H753.zip](https://github.com/Infineon/AIROC-Wi-Fi-Bluetooth-STM32/files/14559609/WiFi_Type_1yn_STM32H753.zip)

npal-cy, 2024-03-18:
Hi, 
@prachichechani 
Unfortunately i do not have Nucleo STM32H753. I have  Nucleo STM32H745ZI. Per https://community.st.com/t5/stm32-mcus-products/stm32h753-and-stm32h743-the-difference/td-p/348630 it should be compatible. 

I try Nucleo STM32H745ZI + 1yn module  and i do not see a problem with timeout. I attached the project which i used. Could you program the elf file to see if it works on your side. 
[wifi_scan__1yn.zip](https://github.com/Infineon/AIROC-Wi-Fi-Bluetooth-STM32/files/14635972/wifi_scan__1yn.zip)

Regards,
Nazar

prachichechani, 2024-03-19:
Hi,
We have implemented both Infineon modules i.e type 1yn & type 1LV with dual core Controller (STM32H747i-DISCO) and it was working properly without any issue.
 
We started implementing STM32H753 + stm32-nucleo adapter  +  type 1LV module. It is working properly without any error. But When we followed the same procedure with STM32H753 + stm32-nucleo adapter  +  type 1YN module, Getting the error as mentioned above.
 
We also checked the difference with the project provided by you and the project created by us for STM32H753 as well as STM32H723, and concluded that both the projects are same. 
 
Looking forward with a positive response.
Thanks & Regards

npal-cy, 2024-03-19:
Hi,
@prachichechani, please confirm that you programed elf from attached project in to your setup Nucleo STM32H753, and behavior the same. Difference between my and your projects are in system clock configuration.

Regards,
Nazar

neilch0811, 2024-03-19:
Hi @prachichechani 

This is Neil Chen, Wi-Fi Prod. Line Marketing in Infineon, can you pls go to our CRM to create a new case for your project? 

And what's the opty for your company? Can you pls drop me a mail so that I can create internal opty and pull resource to support your project. 

My e-mail: neil.chen@infineon.com

Thanks!