Infineon/AIROC-Wi-Fi-Bluetooth-STM32 #17: STM32H747I-DISCO drive murata m.2 LBEE5PK2AE-EVB module

Michael5201314, 2024-06-05:
**Prerequisites**

Can you reproduce the problem reliably?
yes
Did you check current release notes for known issues?
yes
If this is not the latest release, have you checked newer releases? 

**Description**:

A clear and concise description of what the bug is.

**Steps to Reproduce:**

1. connect STM32H747I-DISCO and Murata uSD-M.2 Adapter (Rev B), and LBEE5PK2AE-EVB. 

2. compile AIROC-Wi-Fi-Bluetooth-STM32-release-v1.6.0 examples, such as "wifi_scan"

3. burn wifi_scan_CM4.hex and wifi_scan_CM7.hex 

4. power on and observe console prints.

**Expected Result:**

wifi scan result 

**Actual Result:**

 ******************* WiFi-Scan app ******************* 

    Push blue button or send any symbol via serial terminal to continue...
    (The larger, square blue button)

    CYW43xxx detected 

Timeout while waiting for function 2 to be ready
                                               Failed to initialize Wi-Fi

**Frequency (1/1, 1/50, 1/100):**

100%

**Build/Commit:**

Please specify the release, like 1.4.1. If there is no release, give us the commit for the code you used.

**Target:**

The chip or kit you are working with

STM32H747I-DISCO
Murata uSD-M.2 Adapter
Murata LBEE5PK2AE-EVB

**Host OS and Version:**
Freertos

MacOS, Windows, Linux
Windows

**Compiler:**
code migarate to STM32CubeIDE 1.15.1 

**Environment**

I connect WL_REG_ON_HOST and WL_HOST_WAKE_HOST signals between STM32H747I-DISCO and  uSD-M.2 Adapter,

set SDIO to 3.3V. I can capture SD_CLK = 37.5M on oscilliscope , also SD_CMD, SD_DATA can be seen when power up.

Rakesh-BG, 2024-06-05:
Hi,

Please use Murata uSD-M.2 Adapter (Rev PA2) which will resolve your issue.

Thanks,
Rakesh B G

Michael5201314, 2024-06-06:
Hi Rakesh,

Thank you for your reply. I tried to set the SDIO voltage to 1.8V, and it also failed. Then I switch to murata tyep2EA (Infineon CYW55573), it also failed as follow picture shows:

***@***.***
Is it a software problem or a hardware problem? STM32H747I-DISCO or the m.2 interface board?

BR, Michael